todbot/blink1 #317: Saved pattern does not play when plugged into a Mac

dimitribouniol, 2017-10-16:
If I save a pattern to a MK2 device, it will play when a server tickle expires, when it is plugged into a wall outlet (iPhone USB charger), and when it is plugged into a Windows machine, but not when plugged into a Mac. On a Mac, it simply blinks white really quickly, then turns off until a command is sent to it.

I have tried this on a variety of HW, from Mac minis to MacBook Pros, running on pretty much any version of macOS I could find (including 10.11-10.13).

My specific use case that I'm trying to achieve is to have a color specific to system boot and shutdown, while the blink has power, but has not been sent any commands yet. I'm using the blink1-tool (v. 0.98) to send the commands. The blink device has the following version: `firmware:204 serialnum:2000A3B3 (mk2)`.

Any help would be greatly appreciated. Thank you!

(This is probably related to #234, but I admit I had a much lesser understanding as to why it wasn't working when I wrote that back in 2015)

todbot, 2017-10-16:
Are you running the Blink1Control application or any other type of program that accesses the blink(1) on the Mac?

dimitribouniol, 2017-10-16:
Nope, just the command line tool.

todbot, 2017-10-17:
Apologies I didn't read your original issue very carefully.
The saved color pattern will only play under three conditions:
- The blink(1) is powered but not configured as a USB device (e.g. in a USB charger)
- Explicitly told to play with the `--play` command
- After the `--servertickle` command's timeout expires

So I'm confused as to how it played on the Windows machine, unless the Windows machine was actually asleep but left its USB ports powered.

In normal `--servertickle` use, you will need to issue that command and then wait for the timeout to expire before the pattern will play.  E.g. something like:
1. Insert blink(1) device
2. Run `blink1-tool --rgb=0,255,0` to set initial color of the device
3. Run `blink1-tool -t 5000 --servertickle 1` to start the servertickle timer
4. Keep issuing `blink1-tool -t 5000 --servertickle 1` before 5000 milliseconds (5 secs) is up
5. Stop issuing the `--servertickle` command
6. After 5 seconds, the saved pattern will play

dimitribouniol, 2017-10-18:
> but not configured as a USB device

This seems to mirror my experience. As you suggested, this is how I am using the server tickle (Though with `1,1` - if the second `,1` is not set, then the current color get's cleared out I believe, but I'm not completely sure).

I'm glad I'm not missing anything, though I guess the experience I was hoping for was that if the server tickle never came, including the case where the machine completely turned off (power failure, force shutdown, etc...), then as soon as the blink got power, it would play its pattern until another color was set. I assume such a change would need a firmware overhaul, but would be very useful to indicate a machine is still booting up from a failure condition.

That said, for now I'll modify my incorrect assumptions regarding this and figure out a way to best represent this state :) Thank you!

If there is any documentation regarding the `[,1/0]` section of the `--servertickle` command, that would be very helpful as well - I can't remember what 2-years-ago-me understood about that parameter, but I'm using it all over the place...