todbot/blink1 #320: LED Number not working in python 3

gsilvers, 2017-12-23:
The following code is not working in python 3:

`b1.fade_to_rgb(1000, 255, 42, 0,1)
time.sleep(1)
b1.fade_to_rgb(1000,36,111,201,2)
`

Program does not crash or return any error but both LED's are set not each individually. I am assuming the USB logic is wrong but am checking.

todbot, 2017-12-23:
Hi @gsilvers,
Thanks for looking into this. We're in the process of moving the python module to https://github.com/todbot/blink1-python
I assume you installed via pypi?  Pypi is pulling from that repo.

Could you open that issue there?  I'll also start looking into it too, but holidays are in the way for the next few days.

gsilvers, 2017-12-23:
Sure I can move it. I may have some time sunday to look. Is there any documentation of the strings you are sending via lib USB. I'm guessing that is the culprit.

todbot, 2017-12-23:
Sorta.  The origin is the firmware: https://github.com/todbot/blink1/blob/master/hardware/firmware_mk2/main.c#L462
but you can also see also go by previous known-working code like:
https://github.com/todbot/blink1/blob/master/commandline/blink1-lib.c#L243

The corresponding section in the Python library is:
https://github.com/todbot/blink1-python/blob/master/blink1/blink1.py#L139

todbot, 2017-12-23:
I found the problem. It was in the python code.  I've updated the https://github.com/todbot/blink1-python repo.  You can get the updated version on a git pull of that, or make the change on your own checkout: https://github.com/todbot/blink1-python/commit/3f40a70edc17e7dd3605763e443f4721301071a7

I'll publish an update to pypi in a day or so.

Thanks for catching this!

closed 2017-12-23 by todbot

gsilvers, 2017-12-24:
This issue was moved to todbot/blink1-python#2