todbot/blink1 #321: No IFTTT after 4 years

ArielLeslie, 2018-02-13:
The docs say that more information on IFTTT is "coming soon" four years ago. Creating IFTTT applets does not work (including simple "make it blink" dummy applets). IFTTT interface indicates that applet ran, but Blink1Control2 says "no IFTTT events yet".

todbot, 2018-02-13:
Hi @ArielLeslie,
Perhaps your "IFTTT key" setting between Blink1Control2 and IFTTT blink(1) Service do not match? 

I just tested IFTTT connectivity with an IFTTT Do button  (https://ifttt.com/applets/61956082d-blink-my-blink-1 ) and it works.

What does your IFTTT recipe / applet look like? You should be able to share it so I can make my own copy and test here.

ArielLeslie, 2018-02-13:
ID is copy-pasted
[image]

Rule name matches
[image]

There are never any IFTTT events or Recent Events
[image]

todbot, 2018-02-13:
Hi @ArielLeslie,
That's odd.  I can see that your events occurred on our IFTTT gateway. But Blink1Control2 isn't getting them.  Are you behind some sort of corporate firewall or is the computer you're using locked down in some way?  Normally Blink1Control2 will report if it cannot connect to the IFTTT gateway but perhaps it has a bug in that regard or is being fooled.