todbot/blink1 #327: Blink1 MK2 only 1 LED is activated

rsam70, 2018-03-22:
Platform Windows 10, using c# API.

For diagnose purposes we use solid(blinking) red,use solid(blinking) green,
use solid(blinking) blue, off.

It sometimes happens that only 1 LED is on whilest expected to be Both on.
folowing code snippet
```C#
if (ledColorMode.Mode == LedMode.Blinking)
{
     succes &= client.Blink.setPatternAtPosition(0, 0, 0, pwm, 0);
     succes &= client.Blink.setPatternAtPosition(ledColorMode.Color.R, ledColorMode.Color.G, ledColorMode.Color.B, pwm, 1);
     succes &= client.Blink.PlayPattern(true, 0, 1, 0); // on 
 }
else
{
     succes &= client.Blink.PlayPattern(false, 0, 1, 0); // off
     succes &= client.Blink.setRGB(ledColorMode.Color.R, ledColorMode.Color.G, ledColorMode.Color.B);
}
```

rsam70, 2018-03-22:
> Hi,
> Which C# API are you using? I cannot tell from the methods you're using.
> It looks sort of like "Blink1Lib" and not "ManagedBlink1", but I don't see
> a PlayPattern() method in "Blink1Lib".

Hello Tod,

Thx for you reaction.
Btw I think this is a VERY useful application!.

Below is the stack we basically use.

Application.
----------------
blink1-cs.dll (Managed Blink lib)
in here are the calls
*[DllImport(DllName, CallingConvention = CallingConvention.Cdecl)]*
*  public static extern System.IntPtr blink1_open();*

*etc.*
----------------
blink1-lib.dll,
----------------
hid.dll.

We are using vs2017 c++17 C#4.7.1 all latest greatest. (64 bit compilation)

When doing the same  calls *sometimes* only 1 LED is lid. Now the guy who
is in charge of testing
this stuff is getting nervous about this.This is also really confusing for
a service technician.
When this LED is approved by an official commitee then
alot of machinary will be equipped with this LED ( OCE/Canon printers ).

Additional questions
1) Is it possible to program the flash with the following sequence
solid red 2 sec off 2 sec (endless). I can now only do a fade to transition.
2) The initial startup sequence ( fading red to blue to green to white) was
probably fun
to flash that, is there a way to overrule that behaviour?

Thx in advance

Ruud Sampers

laurencedobson, 2019-06-04:
I am currently experiencing this problem. My app is using the Blink1 C# API and hidlibrary v3.2.46 in a .NET 4.6.1 app.

The issue occurs from a fresh start-up or after resuming from sleep* with the device already connected. From this point on, calls to SetColor() still affect both LEDs but, calls to PlayPreset() affect only 1 LED. The issue is resolved when removing and re-attaching the Blink(1) device.

I suspect that the "demo mode" (blue, red, green cycle), which appears to occur when the device is plugged in while not in communication with the OS/driver, might be the cause of the issue... perhaps it leaves the device in a state which the C# API doesn't reset/clear.

todbot, 2019-06-04:
Hi @laurencedobson, 
A few questions:
1. Which exact Blink1 C# API are you using? There are a few different ones.  It may be the library can be fixed. 
2. Are you just sending `SetColor()` commands and no others?
3. Does the problem persist if you change your `SetColor()` calls to `FadeToColor(10,...)`?  (This should be visually the same)
4. Are there any other apps running trying to talk to the blink(1) device?
5. What firmware version is the blink(1) device running? You can find this with `blink1-control.exe --fwversion`

(As for which C# API, I normally suggest people use the [Sleddog.Blink1 on NuGet](https://www.nuget.org/packages/Sleddog.Blink1/) just because it's 100% .NET (no C library dependency) and it's in a package manager. Second choice would be our ManagedBlink1 library since it's also 100% .NET.)

laurencedobson, 2019-06-05:
Hi @todbot. Thanks for getting back to me.
Answers to your questions:

1. I am using "@ThingM.Blink1 managed Library", according to the Assembly Information "Title"
2. I call SetColor() to set the device to a solid white (idle) color. Then I use SavePreset() to fill up each of the 32 available presets and PlayPreset(0) to make the device flash alerts
3. I'll give this a try let you know shortly
4. No, no other blink(1) apps installed/running
5. My app is running on a number of workstations, each with their own device - and each exhibiting the same symptoms. The particular device that I am testing on is reported as:
id:0 - firmware:204 serialnum:XXXXXXXX (mk2)

I note that, when using Blink1Control.exe, I everything works as expected. When I then exit and run my app instead, the symptoms still persist.

laurencedobson, 2019-06-05:
Follow-up for 3: Substituting SetColor() with FadetoColor() appears to have made no difference - either solution lights both LEDs correctly and calling SavePreset() and PlayPreset() still exhibits lighting on a single LED.

laurencedobson, 2019-06-05:
While poking around with the command-line tool, blink1-tool.exe, I have found that writing my pattern to the flash memory (overwriting the default alternating pattern) prevents the issue from occurring on future system start-ups. This seems to suggest that saving a pattern to the device which contains led switching might be the root cause, presumably because the .NET library doesn't specify which LED to light when writing patterns to it.

laurencedobson, 2019-06-05:
On a device exhibiting this issue, the pattern being written to it is different. Here's what I did...

1. After start-up, the app sets the pattern.
2. I inspect the pattern stored on the device by use of: blink1-tool --readpattern

The output is:
{0,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1,#000078,0.50,1,#00c832,0.50,1}

The pattern uses 1 (which denotes LED 1 only)

3. I then unplug the device and the app sets the pattern again.
4. This time, I get a different pattern when I repeat: blink1-tool --readpattern

Output:
{0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0,#000078,0.50,0,#00c832,0.50,0}

This time, the pattern is set to uses 0 (denoting both LEDs).

laurencedobson, 2019-06-05:
Additionally, while exhibiting the issue, using blink1-tool.exe to play the pattern on a specific ID results in the pattern playing on LED 1, regardless of which LED I specify. Therefore, each of the following commands produces the same result:
```blink1-tool.exe --play 1,0 -l 0
blink1-tool.exe --play 1,0 -l 1
blink1-tool.exe --play 1,0 -l 2```

laurencedobson, 2019-06-05:
I realised that what I needed was a way to specify the LED via the .NET library, so...

1. I took a look in the /hardware folder in this repository, specifically `hardware/firmware_mk2/main.c` (https://github.com/todbot/blink1/blob/master/hardware/firmware_mk2/main.c)
2. I found the "Available Commands" comments section @ line 476
3. This informed me of the command that I would need, `Set ledn` @ line 486
4. I wrote an additional function, `SetLED()` in `blink1\blink1.cs` which does the necessary `hidlibrary` communication with the blink(1) device
5. Then I modified the `SavePreset()` function to take an additional `ledn` parameter and call the new `SetLED()` function prior to saving the preset to the device

Now, ensuring the device was still exhibiting the issue, I then ran the new code to set the pattern and play it... AND IT WORKED :)

I'll get the changes put into a PR for you.