todbot/blink1 #331: URL parsing not working

gwatkins1957, 2018-04-03:
When I have an event source that points to a text file that is either JSON or plain text format. The parser fails , even when using the examples. JSON parser error is "Unexpected token in JSON at position 0". If I change to plain text then it will show "no color found" or "no pattern found". Here is sample URL with JSON format. https://remote.digitalspeech.net/dss/blinker/133.txt

todbot, 2018-04-03:
Hi,
It looks like that URL is outputting text as perhaps UTF-16 but the server is saying it's text-plain (ISO8859-1).

For instance, try that URL here https://www.hurl.it/ or https://jsonformatter.curiousconcept.com/

In the first you'll see it outputs apparently garbage, but notice it says the content-length is 42 bytes for a 21-byte string.  This is the hallmark of UTF-16 or similar encoding.  The second just fails because it sees a bunch of control codes instead of a json string.

What are you using to generate that string?  It seems like it's not being down-converted to plaintext. (I've seen this happen with certain Java APIs)

(I guess my email response didn't get you yesterday)

gwatkins1957, 2018-04-11:
I actually found the answer before I saw this reply but you were correct. I had to change the character encoding type to ANSI when I was creating the file.

closed 2018-04-11 by gwatkins1957