todbot/blink1 #334: Blink(1) mk2 Device w/Buildroot | "no blink(1) devices found" error

lukemk1, 2018-04-11:
Hello,

I had previous bought one of your blink(1) mk2 devices for testing with an embedded system that I am working on a proof-of-concept for. 

I am utilizing Buildroot ([link](https://buildroot.org/)) for creating my embedded system image, with Linux kernel version Linux Kernel 4.4.43-v7, and have attempted to get it working.

I am using the pre-compiled raspberry pi blink1-tool executable (found [here](https://github.com/todbot/blink1#blink1-tool)), but am unable to get the LEDs to trigger. 

I am continually greeted with the "no blink(1) devices found" error. I have used the following command, both with and without "sudo" prefixing it, just to do a quick test:

`(sudo) ./blink1-tool-rpi -t 400 --random 100`

I can see the usb device via "lsusb" (Bus 001 Device 005: ID 27b8:01ed), but it appears that the physical device and application are not linking up.

I have been able to get the blink(1) device working with regular Ubuntu and Raspbian images (I tested this in hopes to glean any information I could, unfortunately this did not get me anywhere). 

If you have an ideas how I would be able to get the blink(1) to function on under an embedded buildroot device that would be greatly appreciated. 

I could be missing a step somewhere along the way, so I would appreciate your thoughts.

Thanks.

todbot, 2018-04-16:
Hi @lukemk1,
A few questions: 
- What device are you running your buildroot Linux on?
- Can Linux see the device with `lsusb`?
- What does the system log print out when blink(1) is inserted?

Assuming all of the above suggest that your host hardware and Linux is seeing the device just fine, I am thinking you will need to recompile `blink1-tool` for your system.  There is enough variability in device drivers and shared libraries among Linuxes that when binary-compatible, they aren't device-driver compatible.  

Are there steps I could follow to recreate the buildroot you're making?  

Not related but could affect you: I'm in the process of moving `blink1-tool` and its attendant C-library to its own repo: https://github.com/todbot/blink1-tool 
You will see the "commandline" directory disappear in the next few days to be replaced by a submodule link to the `blink1-tool` repo.

lukemk1, 2018-04-16:
Hi @todbot, 

- What device are you running your buildroot Linux on? **Raspberry Pi 3 Model B**
- Can Linux see the device with lsusb? **Yes, as noted above: Bus 001 Device 005: ID 27b8:01ed**
- What does the system log print out when blink(1) is inserted? **Log is as follows:**
```Apr 16 19:35:59 kernel: [ 2206.817822] usb 1-1.4: new full-speed USB device number 4 using dwc_otg
Apr 16 19:35:59 kernel: [ 2206.956928] usb 1-1.4: config 1 interface 0 altsetting 0 has 2 endpoint descriptors, different from the interface descriptor's value: 1
Apr 16 19:35:59 kernel: [ 2206.985306] usb 1-1.4: New USB device found, idVendor=27b8, idProduct=01ed
Apr 16 19:35:59 kernel: [ 2206.995892] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 16 19:35:59 kernel: [ 2207.006836] usb 1-1.4: Product: blink(1) mk2
Apr 16 19:35:59 kernel: [ 2207.014630] usb 1-1.4: Manufacturer: ThingM
Apr 16 19:35:59 kernel: [ 2207.022236] usb 1-1.4: SerialNumber: 2000DE5B
Apr 16 19:35:59 kernel: [ 2207.046593] thingm 0003:27B8:01ED.0001: hidraw0: USB HID v1.01 Device [ThingM blink(1) mk2] on usb-3f980000.usb-1.4/input0```

I could attempt to add the source and compile it with the other buildroot processes if nothing else would work. I was just hoping to avoid that since it would add some complexity. 

I can provide you with the defconfig I am using for my buildroot configuration. Of course you will be missing some of my specific packages which would have to be deselected, but that shouldn't cause any issues if you want to try it.

Let me know if I can provide you with anything else.

lukemk1, 2018-04-19:
Update:

I have compiled the blink-tool from source into my Buildroot generated embedded system but the blink-tool gets the following error:

`./blink1-tool: line 6: syntax error: unexpected "(" (expecting ")")`

I am going to try compiling the mini-tool and see if that is any better.

todbot, 2018-04-19:
So you are running Raspbian on a Raspberry Pi and building a buildroot Linux kernel to also run on a Raspberry Pi?

lukemk1, 2018-04-19:
No, there is not a standard operating system running. Buildroot is a tool which cross-compiles the entire embedded filesystem image from the given parameters. 

So I'm on my Ubuntu host machine, utilizing the buildroot tool (which cross-compiles everything I give it) to generate the filesystem image.

lukemk1, 2018-04-19:
Okay. It's a bit odd since the cross-compiling works for everything else that I've been creating.

Certainly, I can give you the steps. It will probably take me a day since I'm in-between some other things currently. 

Thanks for your help!

lukemk1, 2018-05-03:
Hi @todbot,

I wanted to ping you to see if you have been able to test this out or need any further information from me.

Thanks!

lukemk1, 2018-06-29:
Hi @todbot,

It's been awhile so I wanted to ping you to see if you have been able to test this out or need any further information from me.

Thanks!