todbot/blink1 #337: Permissive license - BSD? MIT?

nightst4r, 2018-07-26:
This project seems really cool and vital to Blink1.  If this is the offical SDK for it, it would be great to see this offer a BSD or MIT style license as well.  The CC-BY-SA is not bad, but there is some ambiguity that might leave people not comfortable using it, and reverting to just using the command line tools and shell exec.

HenkKalkwater, 2020-11-06:
I don't have anything  against a copyleft license, but using the Creative Commons Attribution-Sharealike 3.0 license (CC-BY-SA-3.0) makes the code hard to use the library code in other applications. CC-BY-SA-3.0 is somewhat ambiguous for code, especially the Share Alike part. 

Right now I'm considering linking with the C library within `blink1-tool` (nowadays part of another repo). Does statically linking my code to the blink1-tool count as creating an adaptation of that library that has therefore to be shared under the CC-BY-SA-3.0? Does dynamically linking count as creating an adaptation as well? Does having a program start `blink1-tool` count as creating an adaptation too?  The Creative Commons are a great license for certain artistic works, but they don't provide that much clarity for these things on code.

My suggestion would be to license this work under the CC-BY-SA-4.0. While it is very similar to the CC-BY-SA-3.0, a major difference for software is that [the CC-BY-SA-4.0 is one-way compatible with the GPLv3.0](https://creativecommons.org/share-your-work/licensing-considerations/compatible-licenses). The GPLv3.0 is made to be used by software, similar in intent as the CC-BY-SA in general and is very clear in what counts as a derivative of software and what isn't. Besides that, the GPLv3.0 is more widespread in software communities, making it easier to work with.

I wouldn't mind a permissive license either but if the intent of the license was to prevent people making closed-source software around this product, I can understand that this would not be an option.

todbot, 2020-11-06:
Changing to CC-BY-SA-4.0 sounds like a great idea. I'll start making the changes now.  I think 4.0 wasn't quite ready yet when I last seriously considered licensing.

What's the project you're using blink(1) with?  (I'm always curious where they end up. We rarely see end uses because most seem to be for internal projects)

HenkKalkwater, 2020-11-06:
Currently I'm writing a wrapper library for the D programming language, which I want to use with a simple monitoring daemon I'm planning to write on my Raspberry Pi for fun. It will probably report when players join/leave one of my game servers, failing system services and, if I succeeds, notifications from my Linux desktop. I'm planning to open source when they're actually written :)

GhostTap, 2020-11-06:
I still feel like the official cli should be permissive bsd I can understand the GUI being non permissive. But someone buying the hardware should have a way to use it commercially and package it up if they wanted to. 

If not the tech spec to read and write to the device should be public so people can make their own way of communicating to the device. 

Thanks!

todbot, 2020-11-06:
Hi @GhostTap, I don't see how the current license prevents someone buying the hardware and using it commercially, packaged how they want.  In fact we have several customers that do just that.

GhostTap, 2020-11-06:
It works fully stand alone. But if u wanted to support this software in your game or in another application gpl is not compatible as it would make ur entire software gpl. 

LGPL is closer but most places won’t allow their developers to use that either. 

MIT and BSD are super flexible for adding support for a blink1 into a game or into a software product.

todbot, 2020-11-06:
I'll look into dual-licensing for blink1-tool and similar API-level aspects.

In the past when this has come up, we've done a dual-licensing directly with the customer, but it could be good to make it more public.