todbot/blink1 #340: IFTTT not triggering light

ITCMD, 2018-08-21:
when sending a request to ifttt it triggers in the log ( [https://s8.postimg.cc/dxvwhryj9/ifttt.png](https://s8.postimg.cc/dxvwhryj9/ifttt.png )) but the light never comes on as apposed to controlling it directly from the controls. I am running Linux Mint 19.

todbot, 2018-08-21:
Hi Lucas,
Did you not get my response to your email you sent to the blink(1) contact form earlier today?

todbot, 2018-08-21:
I'll reiterate it here in case my response got caught in a spam folder:

> The "name" in of the IFTTT event source you create in Blink1Control2  must match the "Rule name" in the IFTTT.com blink(1) action service.  
> 
> The fact that you're seeing them in the Blink1Control2 log is good, it means that Blink1Control2 can get on the Net (which is a problem in some corporate settings). So if you make the names match things should start working.
> 
> Let me know if you have any other questions.
> The "name" in of the IFTTT event source you create in Blink1Control2  must match the "Rule name" in the IFTTT.com blink(1) action service.  
> 
> The fact that you're seeing them in the Blink1Control2 log is good, it means that Blink1Control2 can get on the Net (which is a problem in some corporate settings). So if you make the names match things should start working.

ITCMD, 2018-08-21:
I apologize I hadn't received it last I'd checked but I have now. Still, the names seem to match up. Can you take a look at this image? http://i.imgur.com/HkKddK6.png

ITCMD, 2018-08-21:
thanks for your fast response though!

ITCMD, 2018-08-22:
I figured it out. When you said _event source_ I thought you meant pattern. I forgot about the second tab. Thanks for your patience! Ill remove the image since it has my ifttt key.

closed 2018-08-22 by ITCMD

todbot, 2018-08-22:
Awesome, glad you were able to figure it out.
I've never been entirely happy with the tabbed interface. This is a great example of how we should rethink it.
Let me know if you have any other questions!