todbot/blink1 #341: Has v1.98a been retagged?

fxcoudert, 2018-09-10:
Homebrew is shipping `blink1`, but it currently fails to build because the recorded commit ID for `v1.98a` is `de6c0a951af253cb4827604ef3ae89b1643efe28` (which was correct as of 2017-11-26). However, now that tag https://github.com/todbot/blink1/tree/v1.98a appears to be commit ID https://github.com/todbot/blink1/commit/6d3922d737c8a767cc4c58d2b99f0a355dd98c0b and thus Homebrew fails (https://github.com/Homebrew/homebrew-core/issues/31779).

Can you confirm that there was an intentional retag?

fxcoudert, 2018-09-28:
ping @todbot

todbot, 2018-10-06:
PR submitted to homebrew-core.  Thanks for pointing this out.
https://github.com/Homebrew/homebrew-core/pull/32775