todbot/blink1 #344: Blink(1) gets slower when application is minimized

angelrcastaneda, 2019-06-17:
Version: 2.2.2
Electron/Node/Chromium: 4.2.3/10.11.0/69.0.3497.128
------------------------------------------------------------

The Blink(1) USB LED blinks slower when you minimize the Blink1Control2 application.

I'm testing this with a pattern that's 0.1 s on, 0.1 s off, but it also does the same with one of the built-in patterns ("green flash").

The issue does not happen if you lose focus on the Blink1Control2 window, only once minimized.

todbot, 2019-06-17:
Hi @angelrcastaneda, 
What OS are you on?

angelrcastaneda, 2019-06-17:
Hey @todbot - I'm on Windows 10 Pro, 64-bit.

I've disabled battery optimization for the app, as well as unchecking "allow Windows to decide when this app can run in the background".

todbot, 2019-06-17:
Hi, 
About how much slower are you seeing?  That is, is it twice as slow or just a little slower? There's two different potential issues and I'd like to know which to focus on first.

todbot, 2019-06-17:
Ah nevermind, I think have I have a recreation of it on Windows.  Doing some tests now...

todbot, 2019-06-18:
Hi I'm closing this issue since it's about Blink1Control.  
Further effort at https://github.com/todbot/Blink1Control2/issues/115
But I think I have fix.  Thanks for bringing this to my attention.

closed 2019-06-18 by todbot

angelrcastaneda, 2019-06-18:
@todbot That's great - Let me know if there's anything I can provide.

todbot, 2019-07-02:
Hi @angelrcastaneda 
New version is up at: https://github.com/todbot/Blink1Control2/releases
Hopefully that update solves the issue.
Thanks!

angelrcastaneda, 2019-07-02:
Awesome. I'll check it out once I'm back in the office later this week.

angelrcastaneda, 2019-07-11:
Confirmed working. Thank you.