todbot/blink1 #347: macOS System regional number formatting is affecting blink(1) patterns

atchoo78, 2019-08-18:
#### I discovered this today, after some heavy duty debugging until my head nearly imploded:

If the macOS system language is set to any language with a _comma_ separator (delimiter) 
instead of a _period_ (which is the default in all English languages, as far as I know), adding a new pattern through Blink1Control2 won't work unless you replace _all_ period characters with a comma. 

This can be totally confusing because the comma _already_ separates the different segments of the URL,  but I'm afraid the rabbit hole is far deeper and far more complex 🐇

Even if adding the pattern will succeed (and by that I mean that the repeats, timing and rgb hex is "translated" correctly to their respective places in the response text), it is not possible to play the pattern afterwards, because (as you may have guessed by now)...

##  "/pattern/play" is hard coded to understand **periods** instead of commas, and _that's_ what almost made my head implode when I tried to grasp the inner beast of this logic. 

It's kind of like reverse engineering a **catch 22**

## Examples

This will fail :
```
http://localhost:8934/blink1/pattern/add?pname=blink3green&pattern=3,%2300ff00,1.5,%23000000,0.5
```

One theory I have is that the "." is interpreted as a"full stop" (or something else) by OSX, which messes up the resulting pattern that ends up like this: "3,%2300ff00,1,#5%230000,0,5"

This however, works:
```
http://localhost:8934/blink1/pattern/add?pname=blink3green&pattern=3,%2300ff00,1,5,%23000000,0,5
```
But as I mentioned, it's of no use because the pattern won't play correctly unless you implement some clever string replacing, but that's almost impossible to do, due to the complexity of it all, and LOTS of culprits, the "%23" bit being of them.

Ok, I've done my best in explaining the problem, so I leave it hanging there for the moment. Now you know.

Ta-ta-ta-taaaa  🎺

Andreas

atchoo78, 2019-08-18:
Of course, the most obvious fix is to set the decimal separator to ".", but then you have solved one problem and gained 9.999 new ones (or 9,99 - who knows?)

todbot, 2019-08-19:
Hi @atchoo78,
Thanks for this detailed description!  I think this could solve a problem a German user was having too. I'll be looking at this more this week.

Are you essentially changing the "number separators" setting in System Preferences -> Language & Region -> Advanced...?
[image]

atchoo78, 2019-08-22:
@todbot : Major update following

I just did a couple of tests, just to be on the safe side, and it turns out that (most likely) I was wrong in my assumptions regarding the language & region settings.

I wasn't aware that `/pattern/add` expected a "led" parameter (i.e. which Led to use at the color in the sequence. When I looked at some of the example patterns in Blink1Control2, it suddenly occured to me that all the patterns had this extra "0", "1" or "2" parameter, and I had a sneaking suspicion...

Sure enough, when adding an extra "0" to every color of the pattern, it was added flawlessly regardless of decimal separator. That also explains why it "worked" when I use the "," separator.
Because the "led nr" parameter did get a value, even though I wasn't aware of that at the time. :-) 

Time to update the documentation :-)

PS! I just received my blink(1) mk3 today, and it ROCKS with or without decimals! 👍🏼