todbot/blink1 #644: rust library

dadleyy, 2019-09-27:
Hey there! I've been playing around with the blink led lights for a [few years](https://github.com/dadleyy/beacon.client) on and off now and in that time I've gone from playing around with golang to playing around with rust. I'm still _very_ new to rust, but its been a lot of fun playing around with it so far.

Anyways, I was wondering if there was already an officially supported library in rust for controlling blink(1) devices, and if not, if I could volunteer the [`blinkrs`](https://github.com/dadleyy/blinkrs) crate I've been working on to serve that role.

todbot, 2019-10-05:
Hi @dadleyy,
This is great, thanks!   I would be honored if `blinkrs` becomes the official Rust library for blink(1).  

If it's okay with you, I'll put a link to it in this repo in the places that seem appropriate.  This repo is now mostly a list-of-blink1 resources instead of master repo, which was a remnant of my old SVN days.

dadleyy, 2019-10-05:
Thats absolutely okay with me! I'd even be okay with moving ownership of the repository to a github organization; I'm not sure it exists but if a hypothetical `blink-1`/`blink-1-contrib` organization were exists, I'd be happy to move the code I've written to `github.com/blink-1/blinkrs`.

closed 2019-10-08 by dadleyy

todbot, 2019-10-08:
Hi! Thanks for your patience in my response on this.  I've been away from my coding setup.  I just added a link to the Rust library.  There is a `blink1` organization owned by us but there's nothing in it yet. I'm researching what it would mean to move all the repos under `todbot` to `blink1`.