todbot/blink1 #645: Servertickle doesn't work with laptop docking station

ColeBarazoto, 2019-12-03:
My setup is a DELL laptop with the D5000 Dell docking station. I was expecting the server tickle function to play when I unplug my laptop from the docking station, but that doesn't happen.

todbot, 2019-12-03:
Hi,
A few questions:
- What are you seeing instead?  
- Is the docking station powering the USB port?
- What "servertickle" commands are you issuing?

ColeBarazoto, 2019-12-03:
I see whatever color the Blink(1) was set to before unplugging from the dock. Yes, the dock is powering the USB port. Here is the lines of code where I use the server tickle function:

b1.write_pattern_line(100, '#8A0808', 1)
b1.save_pattern()

b1.server_tickle(enable=True, timeout_millis=2000, stay_lit=True)

I could also be misusing the function.