todbot/blink1 #653: How to switch on a color with blink1-tool permanent

sohamakl, 2020-04-09:
Hi,

it might be easy but I'm totally confused. I just want to switch on a specific color with blink1-tool on my windows laptop. This should be on until I change to another color. When computer is shutdown Blink(1) should be off too.

Does anyone know a solution to this?

todbot, 2020-04-09:
Hi @sohamakl,
If you do `blink1-tool --rgb 255,0,255` or similar, that color will stay on the blink(1) until it is commanded otherwise or the blink(1) is disconnected.  Is this not what you're seeing?  

Note you cannot run blink1-tool and Blink1Control simultaneously, as they'll conflict.

sohamakl, 2020-04-09:
If I do `blink1-tool --rgb 255,0,255` I got the response `set dev:0:0 to rgb:0xff,0x00,0xff over 300 msec` blink(1) switchs on the defined color and some seconds later off again.

todbot, 2020-04-09:
The output is correct.  But the turning off is strange. 
Do you have any other programs that control the blink(1) (like Blink1Control2) running at the same time?
Also, what is the output of the command `blink1-tool --fwversion`

sohamakl, 2020-04-09:
Version is `id:0 - firmware:303 serialnum:38b9be26`. I wouldn't know that other processes are accessing my blink(1).

todbot, 2020-04-09:
Yes, that is odd.  A few more questions:
- Just to verify: you're not running the Blink1Control2 app?
- What OS is this on?  
- How is the blink(1) connected to the computer (e.g. direct, direct via cable, through USB hub, etc)
- Does the blink(1) flash any other colors besides the one you chose when it turns off?

sohamakl, 2020-04-09:
- No Blink1Control2 in systray or task manager found
- I'm using Windows 10 home latest public version
- Direct connected to my laptops 
- When windows is running it's just off until I set a color by blink1-tool, when pc is shutdown blink(1) blink in alle colors one by one

sohamakl, 2020-04-09:
I have tested now on an other machine. Also windows 10. No other blink(1) software than blink1-tools.exe installed or downloaded on this machine but same behavior.

todbot, 2020-04-13:
Hi,  I've been researching this on my end but cannot replicate the issue.
Can you time the time between when you command the blink(1) to be a color and when it turns off?  Is that time consistent?

Even though it seems that you're on the latest blink(1) firmware, perhaps try upgrading it just in case.  You can find a download bundle here: https://github.com/todbot/blink1mk3/releases/download/firmware-v303/blink1mk3_fw_update_fw303-2Aug2019c.zip
The instructions are in the zip bundle and also here: https://github.com/todbot/blink1mk3/tree/master/blink1mk3_fw_update

sohamakl, 2020-04-14:
Hi,

the firmware update has not changed anything. The delay with which blink(1) is switched off again is almost exactly 3 seconds.

In the meantime I have tested on a third PC. On this one it works as expected. 

So now it's time to find out which process turns Blink(1) off again. I tried to find out more with HHD Software Device Monitoring Studio and Wireshark. I see that about every 4 seconds a packet is sent to the Blink(1) device. Unfortunately I am unable to find the responsible process. I have finished processes one by one but I have not yet found the responsible one. 

Do you have an idea how I can find the responsible process?

At the beginning I asked the question how I can make sure that Blink(1) does not blink when the PC is turned off. Can you say something about this?

Thanks for working so hard to find the problem. Excellent support :-)

todbot, 2020-04-14:
Thanks!  

That every 4 seconds packet is good diagnostic info.  This to me sounds like either Windows or a driver (third-party mouse or keyboard driver, for instance) is polling any HID devices and then doing something. 

That make me think that one thing to try is to disable the "USB Selective Suspend" setting in power options.  Details here:
https://www.howtogeek.com/296918/how-to-stop-windows-from-powering-off-your-usb-devices/

The blink(1) device is supposed to be set to not suspend, but maybe Windows changed how they handle this in the last year or so?

Thanks for investigating this. If it becomes an issue for others, your work could help them too.

sohamakl, 2020-04-14:
USB Selective Suspend was truly enabled but changing to disable did not solve my issue.

Measurelet, 2020-05-09:
I have the exact same issue. It seems that when the process that commands the blink1 ends, it takes 3 seconds and then the blink1 shuts off. 

I wondered about it, as I only experienced the error when testing colors using the CLI tool. Then when running a control script in python I would not see the error.

The following python code will have the blink1 turned on for approx 15+3 seconds

from blink1.blink1 import Blink1
import time 
b1 = Blink1()
b1.fade_to_rgb(1000,255, 255, 255,ledn=1)
time.sleep(15)

todbot, 2020-05-09:
Hi @Measurelet, can you describe the OS, machine, and USB connectivity between your script and the blink(1) device?  I'd really like to track down this issue.  Also, which firmware version of blink(1) do you have (`blink1-tool --fwversion`)

Measurelet, 2020-05-10:
Hi
Firmware 303
Windows 10 Home 64bit
MSI: Intel Pentium Silver n5000, 1.1 ghz, 4gb ram

I am using the cable from the box that the blink(1) came in - if that is what you mean ? Experiencing the issue both using the blink1-tool in "cmd" and the Python API. Blink1 turns off shortly after the python process ends. Using the Blink1-tool it is very sporadic, and I completely gave up on using it with *.bat files.

todbot, 2020-05-11:
Hmm, it sounds like either somehow Windows is configured to do power-saving operations on USB and shutting down USB ports, or there's another HID driver that's interrogating the blink(1) in a way that's confusing it.  

So assuming one of those:
1. Do you know if your Windows power management settings are different than default? This is a desktop and not a laptop, correct?
2. Do you have any other USB devices that you needed to install drivers for?  (e.g. a keyboard with fancy extras)

And apologies for not being very clear by "USB connectivity".  I meant the USB path of the blink(1) device connected. For instance, "plugged in directly into PC's front USB jack" or "plugged into monitor's USB and monitor's USB is plugged into PC".  If there's something in the middle that could be powering down the port, it's a big suspect.

Thanks for helping diagnose this.  I would love to a reliable test case here so I wouldn't have to pester you all with questions. :)   If we can figure out the cause, and how your systems differ from the many other systems using blink(1)s, we may be able to either toggle some setting in Windows or modify blink(1)'s firmware to deal with it.

Measurelet, 2020-05-12:
Hi

1. Power management is completely disabled - or as much as win10 allows. It is a desktop PC setup for data acquisition that runs day and night.

2. No fancy USB drivers installed. 

I moved it to a powered USB hub recently, but I believe the problem was there before as well. I will try to test if the problem goes away by connecting directly to the PC. 

I will keep you updated

Best regards
Morten

KuestenKeks, 2026-10-29:
I also have the issue, that my blink(1) turns off after a while :(

I've hooked it up to a Windows Server and want it to show the state of backups. When a backup starts I issue a command to play a yellow blinking pattern. But after a short while the blink(1) turns off again.. And when the backup finishes I issue a `blink1-tool.exe --green` command. But after a short while the blink(1) turns off again..

This usecase is the whole reason I bought the blink(1), it's a bit frustrating it doesn't work :\ Anything I could provide to diagnose this?

todbot, 2026-10-29:
Hi @KuestenKeks,
That sounds frustrating. We've found that certain background tasks on certain computers will do a USB bus scan that the blink(1) will interpret as an off command. Try running these two blink1-tool commands, one after the other. 

```
blink1-tool.exe --setstartup 2
blink1-tool.exe --savepattern
``` 

That should cause the blink(1) to ignore these bus scans. Let me know if that works!

KuestenKeks, 2026-11-04:
Hi @todbot ,

well, it kind of works. It doesn't seem to turn off anymore. But my blinking pattern also doesn't work anymore as soon as --setstartup 2 is issued :( Instead of flashing the light is just permanently yellow. But I'd really like it to blink since I already created a sign that says it would be blinking ^^ (And I need to use a pattern because I want the blink-tool to exit after it told the blink(1) to start blinking)

[image]

Is there any more in depth documentation available, that explains the differences between the startup modes? I also think blink(1) could really use some documentation which provides an overview how the device works internally. Everything besides the basics isn't really that intuitive. And I think having an understanding of how the devices and it's internal state including the flash works might also help to understand how the different commands work.

I really like the idea of the device, getting it do what I want just seem needlessly tricky, even as technical person ^^'
Kind regards!