todbot/blink1 #654: 'gmake install' fails with Error 64 (usage: install) on FreeBSD 12.1-RELEASE-p3

ccammack, 2020-05-05:
```
$ su
Password:
# pkg install -y gcc git gmake libiconv
[...]
# cd /tmp && git clone https://github.com/todbot/blink1-tool
[...]
# cd blink1-tool && gmake install
Building blink1-tool for OS=freebsd BLINK1_VERSION=v2.1.0-freebsd-amd64 USBLIB_TYPE=HIDAPI
Type 'make help' for other build products
Submodule 'hidapi' (https://github.com/libusb/hidapi.git) registered for path 'hidapi'
Cloning into '/tmp/blink1-tool/hidapi'...
Submodule path 'hidapi': checked out '083223e77952e1ef57e6b77796536a3359c1b2a3'
patching hidapi
patch -N < patches/freebsd-hidapi-libusb-hid.c.patch || echo "patch already applied"
Hmm...  Looks like a unified diff to me...
The text leading up to this was:
--------------------------
|--- hidapi/libusb/hid.c.orig   2018-11-14 21:19:07.488832000 -0800
|+++ hidapi/libusb/hid.c        2018-11-14 21:29:55.513507000 -0800
--------------------------
Patching file hidapi/libusb/hid.c using Plan A...
Ignoring previously applied (or reversed) patch.
Hunk #1 ignored at 339.
Hunk #2 ignored at 401.
2 out of 2 hunks ignored--saving rejects to hidapi/libusb/hid.c.rej
done
patch already applied
gcc -DUSE_HIDAPI -I./hidapi/hidapi -I/usr/local/include -fPIC -Wall -std=gnu99 -DBLINK1_VERSION=\"""v2.1.0"-freebsd-"amd64""\" -c hidapi/libusb/hid.c -o hidapi/libusb/hid.o
gcc -DUSE_HIDAPI -I./hidapi/hidapi -I/usr/local/include -fPIC -Wall -std=gnu99 -DBLINK1_VERSION=\"""v2.1.0"-freebsd-"amd64""\" -c blink1-lib.c -o blink1-lib.o
gcc -DUSE_HIDAPI -I./hidapi/hidapi -I/usr/local/include -fPIC -Wall -std=gnu99 -DBLINK1_VERSION=\"""v2.1.0"-freebsd-"amd64""\"   -c -o blink1-tool.o blink1-tool.c
gcc -DUSE_HIDAPI -I./hidapi/hidapi -I/usr/local/include -fPIC -Wall -std=gnu99 -DBLINK1_VERSION=\"""v2.1.0"-freebsd-"amd64""\" -c blink1-tool.c -o blink1-tool.o
gcc -DUSE_HIDAPI -I./hidapi/hidapi -I/usr/local/include -fPIC -Wall -std=gnu99 -DBLINK1_VERSION=\"""v2.1.0"-freebsd-"amd64""\" -static ./hidapi/libusb/hid.o blink1-lib.o -lusb -lrt -lpthread -L/usr/local/lib -liconv blink1-tool.o -o blink1-tool
gcc -shared -o libblink1.so -lusb -lrt -lpthread -L/usr/local/lib -liconv -DUSE_HIDAPI -I./hidapi/hidapi -I/usr/local/include -fPIC -Wall -std=gnu99 -DBLINK1_VERSION=\"""v2.1.0"-freebsd-"amd64""\" ./hidapi/libusb/hid.o blink1-lib.o -lusb -lrt -lpthread -L/usr/local/lib -liconv
install -D libblink1.so /usr/local/lib/libblink1.so
usage: install [-bCcpSsUv] [-f flags] [-g group] [-m mode] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               [-B suffix] [-l linkflags] [-N dbdir]
               file1 file2
       install [-bCcpSsUv] [-f flags] [-g group] [-m mode] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               [-B suffix] [-l linkflags] [-N dbdir]
               file1 ... fileN directory
       install -dU [-vU] [-g group] [-m mode] [-N dbdir] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               directory ...
gmake: *** [Makefile:570: install-lib] Error 64`
```
I was able to work around it by installing manually:
```
# gmake && install blink1-tool /usr/local/bin
[...]
```

todbot, 2020-05-05:
Hi,
Thanks for the tips.  I'm not very experienced with FreeBSD. Do you know what Error 64 is?  Is it failing because `/usr/local/lib/` doesn't exist on your system?

ccammack, 2020-05-05:
No, it looks like `/usr/local/lib` definitely exists.

From the [install man page](https://www.freebsd.org/cgi/man.cgi?query=install), it looks like `[-D destdir]` is expecting a destination directory rather than a file on FreeBSD but I'm not sure what that option does.

ccammack, 2020-05-06:
[This](https://github.com/todbot/blink1-tool/pull/27) seems to fix it.

closed 2020-10-07 by ccammack