todbot/blink1 #656: udev rule seems to not work on Pi OS 2020-08-20

jtyree, 2020-09-13:
I added a script to the end of the udev rule to make sure it was getting executed on plug, and I confirmed it is. Even though this rule is getting executed, I can still see the blink device only with sudo. I know this used to work in older versions of the OS. Any chance the change in Linux kernel to 5.4.51 in the 2020-08-20 version of Pi OS broke something?

# Copy this udev with "sudo cp 51-blink1.rules /etc/udev/rules.d/"
# When done, do "sudo udevadm control --reload && sudo udevadm trigger"
# Edit it to suit your type of Linux. It's currently set up for modern Ubuntu
ATTRS{idVendor}=="27b8", ATTRS{idProduct}=="01ed", MODE:="666", GROUP="plugdev", RUN+="/bin/blink_plugged.sh"

todbot, 2020-09-13:
Hi,
Can you run `blink1-tool` as a non-`root` user without the `RUN+=`?  That would mean the udev rule works and it's something with udev vs script running.

If so, I suspect the problem may be that there is a delay between USB plug insertion and when it's available via the USB HID subsystem.  This can take a second or two.  I'm sure there's a way around this in udev, but as test, try adding a `sleep 2` at the top of your `blink_plugged.sh`.

jtyree, 2020-09-13:
I should have included the script (below) - it doesn't actually run the tool, it just logs an entry to confirm that the rule was executed.

`#!/bin/bash
echo "Blink1 plugged  at $(date)" >>/tmp/scripts.log`

blink1-tool only sees the device when it runs as root, regardless of udev.

It later occurred to me that it's probably related to [this](https://github.com/todbot/blink1-tool/issues/28). Because the last time I installed it would have been at v2.0.5, which worked fine on the Pi.

I will verify when I'm back where I can try it.

todbot, 2020-09-13:
Also what version of blink(1) was for?  (mk1, mk2, mk3)

jtyree, 2020-09-13:
I have a mk2 and mk3, so I plugged in both.

with v2.0.5

`
pi@rosey:~/Downloads/blink1-tool-v2.0.5-linux-armv7l $ ./blink1-tool --version
blink1-tool version: v2.0.5-linux-armv7l
pi@rosey:~/Downloads/blink1-tool-v2.0.5-linux-armv7l $ ./blink1-tool --list
no blink(1) devices found
pi@rosey:~/Downloads/blink1-tool-v2.0.5-linux-armv7l $ sudo ./blink1-tool --list
blink(1) list: 
id:0 - serialnum:20008403 (mk2) fw version:204
id:1 - serialnum:38285744 (mk3) fw version:303
pi@rosey:~/Downloads/blink1-tool-v2.0.5-linux-armv7l
`

with v2.1.0

`
pi@rosey:~/Documents/blink1-tool-v2.1.0-linux-armv7l $ ./blink1-tool --version
blink1-tool version: v2.1.0-linux-armv7l
pi@rosey:~/Documents/blink1-tool-v2.1.0-linux-armv7l $ ./blink1-tool --list
no blink(1) devices found
pi@rosey:~/Documents/blink1-tool-v2.1.0-linux-armv7l $ sudo ./blink1-tool --list
no blink(1) devices found
pi@rosey:~/Documents/blink1-tool-v2.1.0-linux-armv7l $
`

and udevadm monitor when plugging first the mk2 then the mk3

`
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[747.134161] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4 (usb)
KERNEL[747.138162] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0 (usb)
KERNEL[747.153182] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005 (hid)
KERNEL[747.161119] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/hidraw/hidraw2 (hidraw)
KERNEL[747.161284] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:red:led0 (leds)
KERNEL[747.161415] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:green:led0 (leds)
KERNEL[747.161542] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:blue:led0 (leds)
KERNEL[747.161663] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:red:led1 (leds)
KERNEL[747.161784] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:green:led1 (leds)
KERNEL[747.161909] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:blue:led1 (leds)
KERNEL[747.162193] bind     /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005 (hid)
KERNEL[747.162443] bind     /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0 (usb)
KERNEL[747.162715] bind     /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4 (usb)
UDEV  [747.216996] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4 (usb)
UDEV  [747.227887] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0 (usb)
UDEV  [747.232889] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005 (hid)
UDEV  [747.242014] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:red:led0 (leds)
UDEV  [747.242301] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:green:led0 (leds)
UDEV  [747.255136] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:green:led1 (leds)
UDEV  [747.256166] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:blue:led0 (leds)
UDEV  [747.257792] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:blue:led1 (leds)
UDEV  [747.259183] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/leds/thingm2:red:led1 (leds)
UDEV  [747.260837] add      /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005/hidraw/hidraw2 (hidraw)
UDEV  [747.265305] bind     /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4:1.0/0003:27B8:01ED.0005 (hid)
[... cut ...]
UDEV  [752.953683] bind     /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.2 (usb)
`

and where the udev rule ran the script (looks like 3 for each part)

`
pi@rosey:/tmp $ cat scripts.log
Blink1 plugged  at Sun Sep 13 16:35:23 EDT 2020
Blink1 plugged  at Sun Sep 13 16:35:23 EDT 2020
Blink1 plugged  at Sun Sep 13 16:35:23 EDT 2020
Blink1 plugged  at Sun Sep 13 16:35:29 EDT 2020
Blink1 plugged  at Sun Sep 13 16:35:29 EDT 2020
Blink1 plugged  at Sun Sep 13 16:35:29 EDT 2020
`

todbot, 2020-09-13:
So this is looking like a problem with hidraw and the hid-led kernel driver?

jtyree, 2020-09-13:
I apologize, I should have started with the goal. I'm using the blink1 in python 3. Everything works great from Windows 10, but when I run the same code on a Pi 4 with the latest OS, it doesn't find any devices. I switched to using the blink1-tool to diagnose the problem and found that it also couldn't find the devices unless it was run as root. I had set everything up last year on a different Pi, including the udev rules, and everything worked perfectly (no root needed), so I was trying to figure out what changed since last year. I came up with the Pi OS and blink version. My Linux experience is shallow, so I'm learning as I go along. I just want to make the LED blink from Python on the Pi like it does so effortlessly from Win 10.

todbot, 2020-09-13:
Gotcha.  Okay it sounds like that for whatever reason hidraw and the `hid-led` driver (that can grab blink(1)s) has changed.  (I'm running the same Raspian kernel so I'm unclear where the issue is)  We've been having an ongoing issue with certain Linux configurations where the kernel grabs blink(1)s because of that `hid-led` driver in hidraw. You may be hitting this too.

But the solution is to use blink1-tool compile with libusb instead of hidraw.  I believe this is the difference you're seeing between blink1-tool v2.0.5 and v2.1.0.  (There's very little difference between the programs otherwise)

So short answer to continue with your work: use blink1-tool-v2.0.5.

jtyree, 2020-09-15:
I had another Pi 4 with a fresh install of Pi OS 2020-08-20 and followed **only** the blink1-python instructions (not installing blink1-tool because I really didn't need it). That worked perfectly the first time with no issues.

Thanks for all of your help and the quick responses!

closed 2020-09-15 by jtyree

normanr, 2020-09-20:
If you're going to compare v2.0.5 and v2.1.0, please unplug and replug the device after you have tested with v2.0.5 and before you test with v2.1.0. v2.0.5 seems to cause the blink1 to get unbound from the hid driver because it uses the libusb backend, this makes it impossible to use with v2.1.0 without rebinding it, or replugging it.

normanr, 2020-09-20:
If you can still reproduce, it would be useful to see the output of `ls -l /dev/bus/usb/*/* /dev/hidraw*` before and after the blink1 is plugged in. The output of `udevadm info` and `udevadm info -a` for the blink1 hidraw device (or the plain usb device if the hidraw device doesn't exist) might be helpful too.

jtyree, 2020-10-08:
This time I was more methodical while setting up a new Pi 4 with Blink1. It turns out that the Blink1 worked fine until I plugged in a USB 3.0 hub that had some devices connected (a couple of USB to serial adapters, USB camera, USB speaker, and USB keypad). After that was plugged in the blink1 device no longer worked without sudo. (The blink1 was not plugged into the hub - it was plugged directly into the Pi4) It was as if the udev rule no longer had any effect. After doing this, unplugging the USB 3 hub made no difference. Whatever it changed was persistent and blink1 would no longer work without sudo, even after reboots. (It does work with sudo, just not without)

Here's where I plugged in the hub:

[   48.986782] v3d fec00000.v3d: MMU error from client L2T (0) at 0x2621000, pte invalid
[  452.265391] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[  452.528065] usb 2-2: New USB device found, idVendor=2109, idProduct=8110, bcdDevice=90.41
[  452.528085] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  452.528101] usb 2-2: Product: USB3.0 Hub             
[  452.528116] usb 2-2: Manufacturer: VIA Labs, Inc.         
[  452.546145] hub 2-2:1.0: USB hub found
[  452.546458] hub 2-2:1.0: 4 ports detected
[  452.624916] usb 1-1.2: new high-speed USB device number 6 using xhci_hcd
[  452.758289] usb 1-1.2: New USB device found, idVendor=2109, idProduct=2811, bcdDevice=90.40
[  452.758307] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  452.758324] usb 1-1.2: Product: USB2.0 Hub             
[  452.758339] usb 1-1.2: Manufacturer: VIA Labs, Inc.         
[  452.760451] hub 1-1.2:1.0: USB hub found
[  452.760803] hub 1-1.2:1.0: 4 ports detected
[  452.965365] usb 2-2.4: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  453.228287] usb 2-2.4: New USB device found, idVendor=2109, idProduct=8110, bcdDevice=90.41
[  453.228306] usb 2-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  453.228323] usb 2-2.4: Product: USB3.0 Hub             
[  453.228339] usb 2-2.4: Manufacturer: VIA Labs, Inc.         
[  453.232502] hub 2-2.4:1.0: USB hub found
[  453.232818] hub 2-2.4:1.0: 4 ports detected
[  453.334909] usb 1-1.2.1: new full-speed USB device number 7 using xhci_hcd
[  453.480677] usb 1-1.2.1: New USB device found, idVendor=1908, idProduct=2070, bcdDevice= 1.00
[  453.480699] usb 1-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  453.480716] usb 1-1.2.1: Product: USB2.0 Device
[  453.480731] usb 1-1.2.1: Manufacturer: Generic
[  453.480746] usb 1-1.2.1: SerialNumber: 20130100ph0
[  453.490516] usb 1-1.2.1: 1:1: cannot get freq at ep 0x2
[  453.624918] usb 1-1.2.2: new full-speed USB device number 8 using xhci_hcd
[  453.772188] usb 1-1.2.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  453.772197] usb 1-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  453.772203] usb 1-1.2.2: Product: CP2104 USB to UART Bridge Controller
[  453.772210] usb 1-1.2.2: Manufacturer: Silicon Labs
[  453.772216] usb 1-1.2.2: SerialNumber: 01D5C1F3
[  453.805623] usbcore: registered new interface driver usbserial_generic
[  453.805862] usbserial: USB Serial support registered for generic
[  453.808895] usbcore: registered new interface driver cp210x
[  453.809047] usbserial: USB Serial support registered for cp210x
[  453.810565] cp210x 1-1.2.2:1.0: cp210x converter detected
[  453.820291] usb 1-1.2.2: cp210x converter now attached to ttyUSB0
[  453.884909] usb 1-1.2.3: new full-speed USB device number 9 using xhci_hcd
[  454.051106] usb 1-1.2.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[  454.051115] usb 1-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  454.051121] usb 1-1.2.3: Product: FT232R USB UART
[  454.051127] usb 1-1.2.3: Manufacturer: FTDI
[  454.051133] usb 1-1.2.3: SerialNumber: AQ01C0EG
[  454.088256] usbcore: registered new interface driver ftdi_sio
[  454.088291] usbserial: USB Serial support registered for FTDI USB Serial Device
[  454.089586] ftdi_sio 1-1.2.3:1.0: FTDI USB Serial Device converter detected
[  454.089848] usb 1-1.2.3: Detected FT232RL
[  454.094065] usb 1-1.2.3: FTDI USB Serial Device converter now attached to ttyUSB1
[  454.164922] usb 1-1.2.4: new high-speed USB device number 10 using xhci_hcd
[  454.316597] usb 1-1.2.4: New USB device found, idVendor=2109, idProduct=2811, bcdDevice=90.40
[  454.316606] usb 1-1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  454.316613] usb 1-1.2.4: Product: USB2.0 Hub             
[  454.316619] usb 1-1.2.4: Manufacturer: VIA Labs, Inc.         
[  454.318083] hub 1-1.2.4:1.0: USB hub found
[  454.318404] hub 1-1.2.4:1.0: 4 ports detected
[  454.654917] usb 1-1.2.4.2: new high-speed USB device number 11 using xhci_hcd
[  454.864131] usb 1-1.2.4.2: New USB device found, idVendor=1e4e, idProduct=0109, bcdDevice= 0.02
[  454.864152] usb 1-1.2.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  454.864169] usb 1-1.2.4.2: Product: USB Microscope
[  454.864184] usb 1-1.2.4.2: Manufacturer: Etron Technology, Inc.
[  454.909549] usb 1-1.2.4.2: Warning! Unlikely big volume range (=4096), cval->res is probably wrong.
[  454.909570] usb 1-1.2.4.2: [3] FU [Mic Capture Volume] ch = 1, val = -4097/-1/1
[... 47 lines cut ...]
crw-rw-r-- 1 root root    189,  25 Oct  8 14:53 /dev/bus/usb/001/026
[... cut ...]
crw-rw-r-- 1 root root    189,   0 Oct  8 14:32 /dev/bus/usb/001/001
crw-rw-r-- 1 root root    189,   1 Oct  8 14:32 /dev/bus/usb/001/002
crw-rw-r-- 1 root root    189,   3 Oct  8 14:32 /dev/bus/usb/001/004
crw-rw-rw- 1 root plugdev 189,  30 Oct  8 15:02 /dev/bus/usb/001/031
crw-rw-r-- 1 root root    189, 128 Oct  8 14:32 /dev/bus/usb/002/001
crw------- 1 root root    244,   0 Oct  8 14:32 /dev/hidraw0
crw------- 1 root root    244,   1 Oct  8 14:32 /dev/hidraw1
crw------- 1 root root    244,   4 Oct  8 15:02 /dev/hidraw4

todbot, 2020-10-08:
Hi @jtyree,  I'm currently looking into and am diagnosing this. What version of `blink1-tool` are you using and did you compile it yourself?

jtyree, 2020-10-08:
This time, I did not compile or even use the blink1-tool. The only thing that I did was `pip3 install blink1` and execute the udev rules changes [here](https://github.com/todbot/blink1-python). I then created a simple python3 script to cycle the LED. It worked fine until I plugged in the hub. After the hub had been plugged in, (and even unplugged) the script stopped seeing the blink and I had to `sudo pip3 install blink1`. Then my script would work only as `sudo python3 script.py`.

jtyree, 2020-10-08:
By the way, this was from a fresh pi os 2020-08-20 install. No remnants were left over from the comments from a few weeks ago. This was all done from scratch with a new install today.

todbot, 2020-10-08:
What do your current udev rules look like and if they don't have these lines, try adding them:
```
KERNEL=="hidraw*", ATTRS{idVendor}=="27b8", ATTRS{idProduct}=="01ed", MODE="0666", GROUP="plugdev"
SUBSYSTEM=="usb", ATTRS{idVendor}=="27b8", ATTRS{idProduct}=="01ed", MODE:="666", GROUP="plugdev"
```

I'm not convinced that will work though because: 1) current Pi OS may have a new udev setup and b) I'm worried the problem is farther down the stack. I am suspecting the `hid-led` Linux kernel driver has been re-enabled in Pi OS and it does weird things with hidraw (which is what `blink1-python1 uses by default)

jtyree, 2020-10-08:
51-blink1.rules was:
SUBSYSTEM=="usb", ATTRS{idVendor}=="27b8", ATTRS{idProduct}=="01ed", MODE:="666", GROUP="plugdev"

updated to:
SUBSYSTEM=="usb", ATTRS{idVendor}=="27b8", ATTRS{idProduct}=="01ed", MODE:="666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="27b8", ATTRS{idProduct}=="01ed", MODE="0666", GROUP="plugdev"
SUBSYSTEM=="usb", ATTRS{idVendor}=="27b8", ATTRS{idProduct}=="01ed", MODE:="666", GROUP="plugdev"

then rebooted

same result

Not a major problem for me right now, I can work around it by turning off that functionality for now.

Thanks for checking into it. Seems like that USB3 hub caused some persistent issues because all was good before plugging it in.

todbot, 2020-10-13:
@jtyree do you have the exact SKU or ASIN of that USB3 hub so I can attempt to replicate your experiences?

jtyree, 2020-10-13:
Give me a few days to research this a bit more so I can give you the steps from Pi OS clean install along with a SKU.

jtyree, 2020-10-16:
I'm having a hard time explaining this one. Here are the steps that I followed. It appears to be not the hub, but another device that happened to be plugged into the hub.

1. Install Pi OS 08-25
2. pip3 install blink1
3. update udev rules per instructions in the blink1-python repo
4. create python test script to blink led (copied from github readme)

  import time
  from blink1.blink1 import blink1

  with blink1() as b1:
    b1.fade_to_color(100, 'navy')
    time.sleep(10)

5. non sudo test with blink1 plugged in USB2 port on Pi - Pass
6. non sudo test with blink1 plugged in USB3 port on Pi - Pass
7. Plug in Gearmo USB3 hub to USB3 port on Pi
8. Plug blink1 in Gearmo hub
9. non sudo test python script - still passes!
10. start plugging other devices in USB3 hub...
11. plug in usb numeric keypad - python blink1 script still works
12. plug in usb [grbl motion controller](https://openbuildspartstore.com/blackbox-motion-control-system/) - now the python blink1 script fails - no longer detects blink1!
13. sudo pip3 install blink1
14. sudo run the same blink1 python script - this works - blink1 is detected with sudo, so I know that the cable and physical connection is good
15. unplug the grbl motion controller to see if the problem goes away - script still doesn't work non sudo. script still does work with sudo.

Everything went bad after the grbl usb device (FT232RL usb-serial bridge chip) was plugged for /ttyUSB0 at  [ 565 ] in dmesg below.

So I reflashed Pi OS again and this time skipped the USB3 hub and plugged the grbl controller directly into the Pi to see if it would have the same result. Everything worked perfectly. 

Then I plugged the USB3 hub back in. Everything was still fine.

Then I plugged the grbl usb device directly into the Pi. Everything still worked.

Then I plugged the grbl usb device into the USB3 hub instead of the Pi. This made it break and stay broken! Blink1 now can only be accessed through sudo. Unplugging the grbl USB device from the hub makes no difference.

**I don't think this is your problem. Looks like I can make it work as long as I don't plug that FT232RL based device into a USB3 hub and instead plug it in directly to the Pi.  I apologize for taking your time on this - if anything, that other device seems to be the one that's causing the problems.**

dmesg logs below

[  131.031662] v3d fec00000.v3d: MMU error from client L2T (0) at 0x7fa1000, pte invalid
[  353.754648] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[  353.894301] usb 1-1.3: New USB device found, idVendor=27b8, idProduct=01ed, bcdDevice= 1.01
[  353.894321] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  353.894337] usb 1-1.3: Product: blink(1) mk3
[  353.894352] usb 1-1.3: Manufacturer: ThingM
[  353.894366] usb 1-1.3: SerialNumber: 37f3f1db
[  353.956733] hid-led 0003:27B8:01ED.0005: hidraw2: USB HID v1.11 Device [ThingM blink(1) mk3] on usb-0000:01:00.0-1.3/input0
[  353.958893] hid-led 0003:27B8:01ED.0005: ThingM blink(1) initialized
[  369.209911] hid-led 0003:27B8:01ED.0006: hidraw2: USB HID v1.11 Device [ThingM blink(1) mk3] on usb-0000:01:00.0-1.3/input0
[  369.210683] hid-led 0003:27B8:01ED.0006: ThingM blink(1) initialized
[  381.627495] usb 1-1.3: USB disconnect, device number 4
[  383.715084] usb 1-1.1: new full-speed USB device number 5 using xhci_hcd
[  383.855228] usb 1-1.1: New USB device found, idVendor=27b8, idProduct=01ed, bcdDevice= 1.01
[  383.855248] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  383.855264] usb 1-1.1: Product: blink(1) mk3
[  383.855279] usb 1-1.1: Manufacturer: ThingM
[  383.855294] usb 1-1.1: SerialNumber: 37f3f1db
[  383.867126] hid-led 0003:27B8:01ED.0007: hidraw2: USB HID v1.11 Device [ThingM blink(1) mk3] on usb-0000:01:00.0-1.1/input0
[  383.867888] hid-led 0003:27B8:01ED.0007: ThingM blink(1) initialized
[  396.470633] hid-led 0003:27B8:01ED.0008: hidraw2: USB HID v1.11 Device [ThingM blink(1) mk3] on usb-0000:01:00.0-1.1/input0
[  396.471425] hid-led 0003:27B8:01ED.0008: ThingM blink(1) initialized
[  399.294080] usb 1-1.1: USB disconnect, device number 5
[  439.785833] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[  439.918928] usb 1-1.1: New USB device found, idVendor=2109, idProduct=2811, bcdDevice=90.40
[  439.918948] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  439.918964] usb 1-1.1: Product: USB2.0 Hub             
[  439.918979] usb 1-1.1: Manufacturer: VIA Labs, Inc.         
[  439.920774] hub 1-1.1:1.0: USB hub found
[  439.921093] hub 1-1.1:1.0: 4 ports detected
[... 85 lines cut ...]
[  743.474467] hub 1-1.1:1.0: 4 ports detected
[  743.660293] usb 2-1.4: new SuperSpeed Gen 1 USB device number 5 using xhci_hcd
[  743.923282] usb 2-1.4: New USB device found, idVendor=2109, idProduct=8110, bcdDevice=90.41
[  743.923300] usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  743.923316] usb 2-1.4: Product: USB3.0 Hub             
[  743.923332] usb 2-1.4: Manufacturer: VIA Labs, Inc.         
[  743.927095] hub 2-1.4:1.0: USB hub found
[  743.927315] hub 2-1.4:1.0: 4 ports detected
[  744.029813] usb 1-1.1.3: new full-speed USB device number 13 using xhci_hcd
[  744.179451] usb 1-1.1.3: New USB device found, idVendor=27b8, idProduct=01ed, bcdDevice= 1.01
[  744.179471] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  744.179487] usb 1-1.1.3: Product: blink(1) mk3
[... cut ...]
[  744.441794] usb 1-1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  744.441800] usb 1-1.1.4: Product: USB2.0 Hub             
[  744.441807] usb 1-1.1.4: Manufacturer: VIA Labs, Inc.         
[  744.442955] hub 1-1.1.4:1.0: USB hub found
[  744.443327] hub 1-1.1.4:1.0: 4 ports detected
[  807.034546] hid-led 0003:27B8:01ED.0010: hidraw2: USB HID v1.11 Device [ThingM blink(1) mk3] on usb-0000:01:00.0-1.1.3/input0
[  807.035318] hid-led 0003:27B8:01ED.0010: ThingM blink(1) initialized

todbot, 2020-10-19:
Hi @jtyree, thanks for the detailed description.  What an odd problem. Is the USB hub you're using powered (i.e. has its own power supply) or is it drawing power from the Pi?  Also, are there any other USB devices plugged into the Pi besides the two devices at a time (the blink(1) + grbl motion controller or blink(1) + keypad)?  Also, what is the current rating of the power supply going to the Pi?

I've noticed many power adapters people use to power a Raspberry Pi are just barely able to power the Pi itself, let alone any USB peripherals. I've had this issue myself, as I started to move to a Pi 4 with its higher power requirements. And I've seen downstream hubs glitch out with power dips, either from the Pi or other peripherals, which is sort of what it looks like in your `dmesg` trace where a hub is reconnecting.  

Alternatively, it could be that the grbl motion control (which appears very beefy) and, while self-powered, could present a high-enough load to USB power that it resets the USB bus, causing the hub to re-enumerate.

If you're not bored of this yet, and you're Pi power is good, I would try plugging the blink(1) into a *different* USB hub (or directly into the Pi) and keep the USB3 hub and grbl motion controller together.  Perhaps that would make a difference.

normanr, 2020-10-20:
Would power explain the difference between non-root vs root (via sudo) access? Does unplugging everything and rebooting fix it, or do you have to reimage to fix it? What's the difference in the output of the `ls` commands before vs after it gets broken? I wonder if there's a udev rule somewhere that can't handle the grbl device connected the via hub correctly and it "corrupts" something about the blink(1)'s permissions.

todbot, 2020-10-20:
Power would not explain the root vs non-root, but perhaps I misread, I didn't think @jtyree was having that issue.

jtyree, 2020-10-20:
The GRBL controller does have something odd going on with power. I found that it could back-power the upstream USB devices from it's 24V supply based on how it was designed. I figured this out because I'd go to reboot the Pi and sometimes it would hang until I unplugged the GRBL device. As soon as I unplugged it, the Pi would resume and boot normally. The Pi always ran fine with it plugged in - the problem only occurred at boot. That was the main reason that I moved it to a hub. (The hub fixed this issue) It never occurred to me that this could impact udev rules and how linux handles the devices, but I obviously need to check. I have a USB optical isolator that I can try putting in line and see if that makes any difference. I've been chasing this one in my spare time which is mostly weekends. I'll let you know how it goes. Thanks for all of the ideas!