todbot/blink1 #657: Improve readme

cdisco, 2020-09-28:
The following in the README.md should be considered a "step 1" and not buried in "OS-specific Tips" for Linux
```Unless you will always be running `blink1-tool` as the `root` user, you should
install udev rules to let any user access the blink(1) device.

todbot, 2020-09-28:
Hi, It is the very first thing on the blink1-tool page (after the meta info about what's in the repo), https://github.com/todbot/blink1-tool#os-specific-tips

Or are you referring to something else?

cdisco, 2020-09-28:
That’s my point it is not a “tip “

todbot, 2020-09-28:
Hi,

Could you maybe create a copy of the README describing what you're asking?  The udev stuff is literally the first OS-specific set of instructions.  Are you suggesting not using the word "tip"?

cdisco, 2020-09-29:
Exactly, drop the word “tip” which has always been optional any readme or instruction manual I’ve ever seen

todbot, 2020-10-01:
Thanks for helping make the documentation clearer.

closed 2020-10-01 by todbot