todbot/blink1 #659: blink(1) mk3 behaves strange with blink1-tool

fbenfhzxdt, 2020-11-20:
Hi,

I have 4 blink(1)s connected to one USB swith.
3 of them are mk2 and one is a new mk3 (bought some days ago).

I use blink1-tool.exe v2.1.0 on Windows 10 Pro (20H2) to control the blink(1)s via a PowerShell script.

Unfortunately, the mk3 shows a strange (erroneous?) behavior.
To reproduce the behavior, it does not matter if I execute the commands via script or directly from command line.
Using cmd or PowerShell makes no difference, too.

Please see the examples below for details about the behavior.

**Devices:**
_blink(1) list:
id:0 - serialnum:2001042E (mk2) fw version:205
id:1 - serialnum:20010466 (mk2) fw version:205
id:2 - serialnum:200110D4 (mk2) fw version:205
id:3 - serialnum:3376bb52 (mk3) fw version:303_

**Example 1:** 
Starting condition: all blink(1)s are turned off.
I turn on ID 1 (mk2) with the command:
`blink1-tool.exe --id 1 -q -m 600 -t 1500 --rgb=ff0000`
and then I enable ID 3 (mk3) with this command:
`blink1-tool.exe --id 3 -q -m 600 -t 1500 --rgb=00ff00`
afterwards I want to turn off ID 1 with the command:
`blink1-tool.exe --id 1 --off`
Expected result: only ID 1 turns off
Actual result: _both_ ID1 _and_ ID3 turn off.

**Example 2:**
Starting condition: all blink(1)s are turned off.
I turn on ID 3 (mk3) with the command:
`blink1-tool.exe --id 3 -q -m 600 -t 1500 --rgb=00ff00`
then I turn on ID 1 (mk2) with the command:
`blink1-tool.exe --id 1 -q -m 600 -t 1500 --rgb=ff0000`
Expected result: ID 3 turning on and after the second command, ID1 also turning on, ID 3 still staying turned on.
Actual result: ID 3 (mk3) turns _off_ and ID 1 turns on.

Somehow the mk3 seems to be reacting to commands directed to other IDs and turns off.
If I only use mk2s in the above mentioned examples, everything works as expected. Only the mk3 is behaving so strange.

Help is highly appreciated.

Thank you

todbot, 2020-11-20:
I'm looking into this now. Thanks for the detailed description.

fbenfhzxdt, 2020-11-21:
Hi, 
Thank you for your real fast investigation and response.

In the meantime, I tried ID1 and ID3 directly on the front-side USB2 ports of my PC case, there the behaviour is the same. 
So I assume that it is not related to the USB hub or USB3.

What exactly would be helpful to describe in my setup? Any hint is appreciated.

I hope the following information is somehow helpful.

My system:
- CPU: Core I9-9900K
- Mainboard: ROG MAXIMUS XI HERO (WI-FI) Call of Duty - Black Ops 4 Edition (ASUS AI Suite 3 installed)

other USB devices connected to my PC:
- I have a Razor Blackwidow Elite keyboard and Razer Mamba Elite mouse, therefore Razer [Synapse 3](https://www2.razer.com/eu-en/synapse-3) is installed on my PC.
- Furthermore, I have Windows Hello enabled with [this ](https://www.kensington.com/p/products/data-protection/biometric/verimark-it-fingerprint-key-fido2-webauth-windows-hello-and-windows-hello-for-business-zpSSG8MOXkqsLIe-r_ShkQ/)Kingston fingerprint reader.
- A [Trustkey G310 ](https://www.trustkeysolutions.com/security-keys/goldengate-g310/)is connected to my PC, too.
- DELL U4919DW connected also via USB3, Dell Display Manager application installed (but I assume that the display manager communicates via DisplayPort as it works without a USB connection, too).
- an additional USB2 sound card (Realtek USB2.0 Audio)

Please let me know if I can do anything else to provide helpful information (e. g. USB tracing or sth. similar).

Thank you very much

fbenfhzxdt, 2020-11-21:
Hi,
I tested one of the mk2 and the mk3 on another PC right now.
This PC is also running Windows 10 Pro 20H2.
Unfortunately, same behavior.

_blink(1) list:
id:0 - serialnum:2001042E (mk2) fw version:205
id:1 - serialnum:3376bb52 (mk3) fw version:303_

Commands submitted:
`blink1-tool.exe --id 0 -q -m 600 -t 1500 --rgb=ff0000` --> mk2 glowing red 
`blink1-tool.exe --id 1 -q -m 600 -t 1500 --rgb=00ff00` --> mk3 glowing green
`blink1-tool.exe --id 0 --off` --> both blink(1)s turning off

todbot, 2020-11-21:
Yes, that is very strange.  Are you typing these commands directly into Powershell or running some script?   Is there any other program running that controls the blink(1) devices?

I would maybe suspect you're using an older version of `blink1-tool.exe` that doesn't parse the `--id` command correctly, but you say you're using 2.1.0.

As another experiment, below is a slightly different build of `blink1-tool.exe`.  Please try it and see if it acts differently.

[blink1-tool-wintest-20201120.zip](https://github.com/todbot/blink1/files/5577091/blink1-tool-wintest-20201120.zip)

fbenfhzxdt, 2020-11-21:
When testing, I type the commands directly into a PowerShell or CMD window. But it makes no difference when submitting the commands in a PowerShell script that is executed by Task Scheduler.

I downloaded blink1-tool v2.1.0 yesterday especially to be sure that I use the current version.
Blink1Control2 is also installed on my main PC (but not on the second PC where I also tested), but not running when I use blink1-tool.
I ensured this via task manager.

I downloaded the new version of blink1-tool you provided, but unfortunately no difference in behavior.
I encountered another strange thing when testing the new version:
`blink1-tool.exe --id 1 -q -m 600 -t 1500 --rgb=ff0000`
 --> ID 1 red
`blink1-tool.exe --id 3 -q -m 600 -t 1500 --rgb=00ff00`
 --> ID 3 (mk3) green
`blink1-tool.exe --list`
 --> ID3 (mk3) turning off - even when only submitting the _--list_ command...?
Trying this with blink1-tool v2.1.0 shows the same behavior.

Furthermore, I did another test on my main PC with Blink1Control2 (v2.2.4) and blinkt1control-tool.exe (v2.1.0).
`blink1control-tool.exe --list`
_blink(1) list:
curl_fetch:http://127.0.0.1:8934/blink1/id
id:0 - serialnum:3376bb52
id:1 - serialnum:200110D4
id:2 - serialnum:20010466
id:3 - serialnum:2001042E_
--> blink(1)s have other IDs than in blink1-tool.exe (mk3 is ID 0 and mk2s are ID1-3)
`blink1control-tool.exe --id 1 -q -m 600 -t 1500 --rgb=00ff00`
_deviceId[0]: 1
curl_fetch:http://127.0.0.1:8934/blink1/fadeToRGB?rgb=%2300ff00&time=0.60&ledn=0&id=1,_
--> should turn on red for ID 1 (one of the mk2s), but actually turns on _only red on the mk3_.
`blink1control-tool.exe --id 2 -q -m 600 -t 1500 --rgb=ff0000`
_deviceId[0]: 2
curl_fetch:http://127.0.0.1:8934/blink1/fadeToRGB?rgb=%23ff0000&time=0.60&ledn=0&id=2,_
--> should turn on ID 2 (mk2) to green, but actually _only changes the mk3 from red to green_.

Thank you very much

todbot, 2020-11-21:
It does appear like the mk3 is misbehaving, but I cannot fathom how it could be acting like this.
These kinds of symptoms would software (`blink1-tool`) or OS-level issues, not blink(1), as the software is just issuing very simple USB HID commands and the OS and general architecture of USB prevents any multiple devices from responding to the same command.

Do you have any non-standard device drivers that control HID devices? I could maybe be persuaded that something like an errant RGB keyboard driver is intercepting USB HID commands and altering them.

One other thing to try is addressing the blink(1)s by their serial number instead of their list index. That is, try:
```
blink1-tool.exe --id 2001042E --rgb=ff0000
blink1-tool.exe --id 3376bb52 --rgb=00ff00
```

If you like, we can send you a replacement blink(1) mk3 (or even a replacement blink(1) mk2, we have a few remainders).  Email us at blink1 at thingm.com with your preferred shipping address and we'll get it shipped out next week.  If this one mk3 you have is weirdly defective, the replacement should work fine. Otherwise, I'd start suspecting hardware (USB hubs, etc) and OS/drivers.

I really wish I had a good theory as to what's happening. This is a really unique issue!

fbenfhzxdt, 2020-11-21:
Thank you very much for your proposal 👍 
You have email :blush:

I tested with the serial numbers as ID on my main PC, unfortunately no difference.
`blink1-tool.exe --id 2001042E --rgb=ff0000`
`blink1-tool.exe --id 3376bb52 --rgb=00ff00`

In the meantime, I tested the mk3 on a HP Elitebook 840 G5 (3rd device now) where nothing special except the required HP drivers is installed.
`blink1-tool.exe --id 0 -q -m 600 -t 1500 --rgb=ff0000`
-->mk3 red
`blink1-tool.exe --list`
--> mk3 again turning off
Same strange behavior on the 3rd device I tested so far.
I assume that it is not related to any driver or hardware issue. But I also do not have any idea what could cause this.

Once more, thank you very much for all your help.

sba923, 2025-01-10:
I'm experiencing a very similar problem with my mk3 devices.

I have the following 3 devices:

1. `serialnum:2000F0A3` (mk2) `fw version:204` -- `blink1-tool.exe --version` outputs `blink1-tool version: v2.0.2-windows-x86_64, fw version: 204`
2. `serialnum:37ff9e03`  `fw version:303` -- `blink1-tool.exe --version` outputs `blink1-tool version: v2.0.2-windows-x86_64, fw version: 303`
3. `serialnum:3748e49e`  `fw version:303` -- `blink1-tool.exe --version` outputs `blink1-tool version: v2.0.2-windows-x86_64, fw version: 303`

If I issue:

`blink1-tool --red`

to the mk2 device, it remains red forever.

If I do the same thing with a mk3, the device turns itself off automatically after a few seconds. Both mk3 devices are affected.

The host is a Windows 11 24H2 PC. This occurs with all other similar PCs.

In some cases issuing `blink1-tool.exe –list` causes the mk3 device to turn off.

I can't repro this with a mk3 connected to a Ubuntu 24 PC (my DIY NAS).

Is that an issue with the mk3 firmware? Hardware?

sba923, 2025-01-19:
@todbot how can I help troubleshooting / fixing this?