todbot/blink1 #660: Big Sur Support

ghctim, 2020-11-20:
Under Big Sur I am unable to make this work.  I am using Intel mac (not M1 Mac) but it I also don't see it under USB devices
[image]

todbot, 2020-11-20:
Hi! We've got a new test machine with Big Sur showing up next week.  Hopefully we should have a new release out shortly after that.  Thanks for your patience!

todbot, 2020-11-20:
Also, I don't see the blink(1) in your USB Device Tree screenshot.  The blink(1) USB Vendor Id & Product Ids are 0x27B8 and 0x01ED.  I'm not sure what that "USB Billboard Device" is but it's not a blink(1).  If you have a USB extension cable and/or USB hub plugged in between the Mac and the blink(1), I would try swapping it out.

ghctim, 2020-11-20:
I have it connected direct with an apple usb c to usb a cable

todbot, 2020-11-20:
Hmm, as a test, do you have a USB hub you could plug it into?

ghctim, 2020-11-20:
I have tried that too and it doesn't show up.  do you think this means its fried?  Its served me well over all these years...

todbot, 2020-11-20:
It's rare but it could be dead. Do you have any other computers to plug it into, just to see if it shows up in the device list?

ghctim, 2020-11-20:
I have a windows computer..yuck, lol

ghctim, 2020-11-20:
I didn't get a new device "ding" when plugging it into windows...

todbot, 2020-11-20:
Okay sounds like it died. If it was one of the early blink(1) mk2 units, there's an increased possibility of "early retirement".  Send us at email at blink1 at thingm.com with your preferred shipping address and we'll get a replacement mailed out to you.

ghctim, 2020-11-20:
oh wow! I was not expecting that all! Thank you. this was an early one!

todbot, 2020-11-20:
Closing as I think this is not a Big Sur issue. :)

closed 2020-11-20 by todbot