todbot/blink1 #664: Not able to send POST or GET

two9seven, 2021-12-01:
I am hoping to control blink1 with POST and GET commands. Is this implemented?

todbot, 2021-12-01:
Hi,

Could you provide more details?  What application are you using to control blink(1)?  There are many.  

If this is about Blink1Control2, please file an issue in the [Blink1Control2 repo](https://github.com/todbot/Blink1Control2/).  

If this is about blink1-tool, please file an issue in the [blink1-tool repo](https://github.com/todbot/blink1-tool/).

If this is a general query, there are many ways to control blink(1) with HTTP requests.  One way is with the [`blink1-tiny-server`](https://github.com/todbot/blink1-tool/tree/master/server) command-line program that's part of the `blink1-tool` repo.  Another way is with the built-in [HTTP REST API](https://github.com/todbot/Blink1Control2/blob/main/docs/app-url-api-examples.md) that's part of Blink1Control2 desktop app. 

If you prefer NodeJs, you can try [node-blink1-server](https://github.com/todbot/node-blink1-server).  Many blink(1) customers write their own HTTP-to-blink(1) gateway using the language of their choice and the corresponding blink(1) API.

two9seven, 2021-12-01:
Thanks for the quick reply! I'm using Blink1Control2 with the API server turned on. 
[image]

I am using Postman to test the POST Request which fails. I get a response from a GET request, however. 

I am able to change the light with the Curl commands that are documented.

two9seven, 2021-12-01:
Solved my own problem by using GET. Thanks anyway!

todbot, 2021-12-01:
Awesome!  Let me know if you have any other questions.

closed 2021-12-01 by todbot