todbot/blink1 #670: blink1control2 can't find device

simsus, 2023-03-05:
Hi everyone

Today I have installed blink1control2 (Debian package) in version 2.2.9 on my Ubuntu 20.04 OS.

After starting the application the status was "no device connected".

I run terminal with the prompt "lsusb | grep "blink"

`$ lsusb | grep "blink"`
`Bus 001 Device 011: ID 27b8:01ed ThingM blink(1)`

I found some issue threads in this repository but no solution seems to solve my problem. Maybe.

greets

simsus

todbot, 2023-03-05:
Hi,
If you're on Linux, you'll need to set up udev rules so non-root processes can access blink(1). 

This is described here: https://github.com/todbot/Blink1Control2#linux
and here: https://github.com/todbot/blink1-tool#linux-including-raspberry-pi

simsus, 2023-03-10:
Oh shit ...

Notice to me: RTFM!!!!

I am so sorry!

closed 2023-03-10 by simsus

todbot, 2023-03-10:
😀 Not a problem. It catches every one up. And happens to me every time I install Linux because I forget