todbot/blink1 #675: FeatureReportByteLength of blink(1) is 9 instead of 8

kissfrog, 2023-09-19:
Tried to get caps via [`HidP_GetCaps`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/hidpi/nf-hidpi-hidp_getcaps) API on Windows, and it turned out that `FeatureReportByteLength` is **9**, which is different from the [doc](https://github.com/todbot/blink1/blob/main/docs/blink1-hid-commands.md):

> The blink(1) feature report packet is 8 bytes long.

Why?

todbot, 2023-09-19:
HID reports with Report IDs are one byte bigger than what's in the HID Report Descriptor, in order to hold the Report ID.

kissfrog, 2023-09-20:
> HID reports with Report IDs are one byte bigger than what's in the HID Report Descriptor, in order to hold the Report ID.

But as per your document, the first byte is also already `report_id`:

```
- byte 0 = 0x01  (report_id )
- byte 1 = 'c'   (command "fade to rgb")
- byte 2 = red value
- byte 3 = green value
- byte 4 = blue value
- byte 5 = th    (fadeMillis/10 high byte)
- byte 6 = tl    (fadeMillis/10 low byte)
- byte 7 = ledn  (unused on mk1, 0=all on mk2) 
```

So actually the `byte 8` exists, but always set as 0?

kissfrog, 2023-09-20:
And what's the **HID Report Descriptor** part of blink(1)? From byte 1 to byte 8 (and the byte 8 is omitted)?